Flybrix/flybrix-firmware #3: thrust PID control implementation

rjwalters, 2016-03-09:
The cascaded PID controller for thrust is not really implemented properly yet. It's nice to use the same controller code that we use for our other three control vectors, but the thrust slave loop isn't equivalent to the gyroscope-backed rotation slave loops. We do not have an ascent/descent sensor that offers better bandwidth than the derivative of the barometer data. (It isn't even clear what sensor would work here -- forward optical flow? doppler ultrasound?)

**We should probably explicitly guard against the pidBypass modes that would put us into cascaded control mode for thrust.**

Slave-only mode can still make sense if we use the derivative of the state altitude estimate -- but we would need to special case some of the command code so that our thrust joystick is bi-directional to allow both ascent and descent.

adnanademovic, 2016-03-11:
Should we just do it by filtering the pidBypass mode, so that we ignore bad bypasses, even if they are sent?

rjwalters, 2016-03-11:
We can send a rejection code back too, right?

adnanademovic, 2016-03-11:
Yes. We could make a verification with returned debug data for each subsystem that could detect that something is wrong. That would be a future proof solution.

adnanademovic, 2016-03-14:
This fixes that problem. We can easily add any verification now, so we should probably use this branch to add more verification.

commit 9de94c6c (2016-03-11): Fix #3 Add verification to config

Configurations are now loaded into a temporary variable, prior to being
written into the actual configuration location. Between those two steps,
the configuration is verified for any illegal settings, and failing at
it prevents writing the actual variable.

The verification syntax seems overly complex at the moment, but it
allows us a simple implementation of verification by each subsystem when
the need arises. With this syntax, failure of verification for one
system doesn't stop the verification process, but continues it, so we
can detect all issues, and not [... cut]
closed 2016-07-13 by adnanademovic