Flybrix/flybrix-firmware #6: forced calibration when arming

rjwalters, 2016-03-09:
The default behavior now is to assume that the flyer is level when we arm and to measure and store the orientation of the flight board with respect to the flyer coordinate system. This is intentional to simplify flight for new models, but can be annoying if flying the same model repeatedly, or if the user wants to take off without being flat without having to trim in flight.

An alternative is to store the rotation matrix coordinates and accelBias in the eeprom and create a flag to enable/disable recalibration during the arming sequence. This seems like something we can support without changing the default behavior if enough of our advanced users want the feature -- but it will require bumping the minor version number.

Jukkorsis, 2016-12-31:
Auto calibrating would be great.
My flyer appears (after investigating with the Chrome app) to have a bias toward tilting backwards and rotating clockwise.

peterlada, 2017-01-16:
Mine has the same issue. Rotating clockwise a good amount and tilting back a little.

rjwalters, 2018-02-27:
v1.6.0 moves our state estimation to a new UKF and adds the ability to calibrate the imu from the app.

closed 2018-02-27 by rjwalters